shiffman/OpenKinect-for-Processing #113: How to get and Read depth information from kinect v2

BHAASKARR, 2017-03-31:
Hi EVeryone, 

I am very new to ROS, I installed ROS and also libfreenect2 Driver and iai_kinect2 library (which are required software for the kinect v2 to interface with ROS).

Ubantu 14.04 ROS, libreenect2, Iai_kinect2

I have below data when i use 'rostopic list' command

bhaskar@D125-40806Linux:~$ rostopic list /kinect2/hd/camera_info
/kinect2/hd/image_color
/kinect2/hd/image_color/compressed
/kinect2/hd/image_color_rect
/kinect2/hd/image_color_rect/compressed
/kinect2/hd/image_depth_rect
/kinect2/hd/image_depth_rect/compressed
/kinect2/hd/image_mono
/kinect2/hd/image_mono/compressed
/kinect2/hd/image_mono_rect
/kinect2/hd/image_mono_rect/compressed
/kinect2/qhd/camera_info
/kinect2/qhd/image_color
/kinect2/qhd/image_color/compressed
/kinect2/qhd/image_color_rect
/kinect2/qhd/image_color_rect/compressed
/kinect2/qhd/image_depth_rect
/kinect2/qhd/image_depth_rect/compressed
/kinect2/qhd/image_mono
/kinect2/qhd/image_mono/compressed
/kinect2/qhd/image_mono_rect
/kinect2/qhd/image_mono_rect/compressed
/kinect2/sd/camera_info
/kinect2/sd/image_color_rect /kinect2/sd/image_color_rect/compressed
/kinect2/sd/image_depth
/kinect2/sd/image_depth/compressed
/kinect2/sd/image_depth_rect
/kinect2/sd/image_depth_rect/compressed
/kinect2/sd/image_ir
/kinect2/sd/image_ir/compressed
/kinect2/sd/image_ir_rect
/kinect2/sd/image_ir_rect/compressed
/rosout
/rosout_agg

Can you please tell what each list describes ? I would like to know which one is useful to get depth values of image and also how to read the those depth data ?

when I used below command
rostopic echo /kinect2/sd/image_depth

I got below values

header:
seq: 1298
stamp:
secs: 1489614685
nsecs: 946450545
frame_id: kinect2_ir_optical_frame
height: 424
width: 512
encoding: 16UC1
is_bigendian: 0
step: 1024
data: [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 177, 30, 244, 28, 206, 4, 0, 0, 0, 0, 57, 5, 0, 0, 27, 9, 159, 6, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 114, 4, 0, 0, 0, 0, 0, 0, 34, 37, 0, 0, 0, 0, 42, 9, 238, 32, 0, 0, 0, 0, 0,.................]

Can you please explain what each value represents above and how it is generated ?

Thanks in advance,
Bhaskar.

BHAASKARR, 2017-03-31:
@danbernier 
@shiffman 
@ThomasLengeling 
@gohai 
@nicolasraga1 

Please help me on this

ThomasLengeling, 2017-03-31:
you should checkout ROS and libfreenect implmentation

closed 2017-03-31 by ThomasLengeling

BHAASKARR, 2017-04-03:
ok Thank you @ThomasLengeling