shiffman/OpenKinect-for-Processing #114: No Device Connected! Cannot Find Devices

LaVielle, 2017-03-31:
I am trying to use a Kinect 1520 in Processing 3 on a Macbook Pro (mid 2012, which has USB 3.0 I believe) with OSX 10.10.5, but I get the following error:

```
64 mac
:/Applications/Processing.app/Contents/Java/core/library:/Users/Sam/Documents/Processing/libraries/openkinect_processing/library:/Applications/Processing.app/Contents/MacOS/v2/mac
Found path: /Users/Sam/Documents/Processing/libraries/openkinect_processing/library/v2/mac/
Init Kinect2 
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 9 usb devices connected
[Freenect2Impl] found 0 devices
No Device Connected!
Cannot Find Devices
EXIT
closing Kinect v2 :
```

I did some research and it seems this issue is somewhat common. I installed libfreenect using Homebrew and `freenect-glview` returns:

```
dyld: Library not loaded: /opt/X11/lib/libglut.3.dylib
  Referenced from: /usr/local/bin/freenect-glview
  Reason: image not found
Trace/BPT trap: 5
```

I don't know if this is what is supposed to happen, but anyways, the Kinect never appears to be connected (NuiSensor Adaptor has nothing in it when I check System Information).

So I am wondering: Is this a hardware issue? Is there an existing solution to this problem? Otherwise, what can I do about it to find a way to use this Kinect in Processing?

ThomasLengeling, 2017-03-31:
Hi try to connect several times, sometimes it takes a couple of tries to make it work, I have some progress updating the processing lib with the new libfreenect library, so I will posting the update soon.

LaVielle, 2017-04-03:
Hi Thomas,
I saw another issue which suggested connecting the device several times. I tried at least 10-15 times, restarting Processing too, but no progress, so wondering about how many times I should try before assuming the problem lies somewhere else...

CharlesFr, 2017-04-07:
Same issue on OS X El Capitan 10.11.2 👎

surr0unded, 2017-04-13:
Hi Thomas,
I have the same issue on Windows 10.

ThomasLengeling, 2017-04-13:
Hi, I'll try to make an update with the processing and the new libfreenect2 by today or tomorrow

lowtechism, 2017-04-29:
I had same issue on OSX 10.10.5 Yosemite + Macbook Pro Late 2012 and Macbook Air Mid 2015 + XBox One Kinect v2. Processing version is 3.3.2.
Today I updated Macbook Pro to OSX Sierra 10.12.4 and Kinect is working fine now. fps is around 20fps.

LaVielle, 2017-05-15:
I updated to OS 10.12.4 and added a few components that I realized were missing from my machine (CMake, and a few others).
The Kinect is now recognized, but I get the error described in issue #65 ...

I think we might be able to close this issue, the solution being:

- update Processing
- update OSX
- make sure you have all the required components installed [as described here](https://github.com/OpenKinect/libfreenect2/blob/master/README.md#mac-osx).

ingrydss, 2017-06-21:
hi LaVielle ,
can you describe what are the others components that were missing?

I have the same problem on windows 10 with just this library (open kinect), but my kinect v2 works on xbox and  with another library called kinectPV2.

OmegaNalphA, 2017-08-08:
Check out my thread [here](https://github.com/shiffman/OpenKinect-for-Processing/issues/104) but most likely you will need to download. Check these out too 

https://github.com/shiffman/OpenKinect-for-Processing#windows
https://github.com/OpenKinect/libfreenect2/wiki/Troubleshooting

Although this is solution for windows, it may work for Mac as well

totovr, 2017-11-11:
I could set it with Kinect V1

https://github.com/totovr/Processing/blob/master/README.md