shiffman/OpenKinect-for-Processing #118: how to remove my silhouette in the v1 Pointcloud example

FranciscoG, 2017-06-21:
When I run the [pointcould v1 example](https://github.com/shiffman/OpenKinect-for-Processing/blob/master/OpenKinect-Processing/examples/Kinect_v1/PointCloud/PointCloud.pde) I see my silhouette along with the pointcloud.  Is there a way to just output the pointcloud?

Here's what I'm seeing.   I tried getting farther away and still saw my shadow/silhouette

[image]

proxycase, 2017-07-09:
That's a "feature" of the Kinect. Either move your Kinect to get better coverage of what you want to see in the Kinect view (since the Kinect projects its own light into the space) or use something like the DepthThreshold example to remove the background after a certain depth.

mirceape, 2018-09-02:
hi there, @proxycase . i ran into the same issue and read about the DepthTreshold, which indeed sorts the problem out but i can't seem to figure out a way to integrate it in the PointCloud.pde file:(