shiffman/OpenKinect-for-Processing #119: Does this work with Raspberry Pi?

Sensebellum, 2017-06-22:
Does this library work with Processing 3.0 on the Raspberry Pi? 

So sorry if this is not the right place for such a query.

proxycase, 2017-07-09:
I would say maybe with the Rpi v3, and the Kinect v1. Looks like there's a kernel driver available here too: https://github.com/xxorde/librekinect which ultimately might be more efficient. With the kinect v2 I think it would be difficult or impossible, since it requires so much more computing power... and USB 3.

Sensebellum, 2017-07-11:
Thank you very much for the contribution I will check that out soon!

jlstanus, 2018-02-16:
@totovr V2 needs USB3, no?

totovr, 2018-02-16:
Sorry I made a mistake!

noonana73, 2018-06-27:
@totovr Hey, do you have any tutorials. I'm trying to use Processing and my kinect on my raspberry pie 3 any ideas?

totovr, 2018-06-28:
@noonana73  I have some examples in this [repository](https://github.com/totovr/Raspberry)

noonana73, 2018-06-28:
@totovr Thanks a lot !!

whatmood, 2020-06-05:
@totovr Hey, I saw the examples you said before. do you have any code example which can be run on raspberry pie3 with kinect?  Does this library 'OpenKinect-for-Processing' work with Processing 3.0 on the Raspberry Pi?