shiffman/OpenKinect-for-Processing #127: kinect for windows Failed to set the LED of K4W or 1473 device: -1

sener07, 2017-10-08:
when i wrote lsusb to terminal it gives me this output:
Bus 001 Device 002: ID 8087:8001 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 005: ID 13d3:3410 IMC Networks 
Bus 002 Device 004: ID 0bda:570b Realtek Semiconductor Corp. 
Bus 002 Device 003: ID 0bda:0129 Realtek Semiconductor Corp. RTS5129 Card Reader Controller
Bus 002 Device 017: ID 045e:02bf Microsoft Corp. Kinect for Windows NUI Camera
Bus 002 Device 016: ID 045e:02be Microsoft Corp. Kinect for Windows NUI Audio
Bus 002 Device 015: ID 045e:02c2 Microsoft Corp. Kinect for Windows NUI Motor
Bus 002 Device 014: ID 046d:c080 Logitech, Inc. 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

when i try to test kinect with this code freenect-glview this gives me this output:
Kinect camera test
Number of devices found: 1
Found sibling device [same parent]
Failed to set the LED of K4W or 1473 device: -1
Found sibling device [same parent]
Could not open device: -1
Could not open device

i cannot fix this error, i tried to use openni but i cant install it it gives me error. Can you help me please thank you.

sener07, 2017-10-10:
$ freenect-camtest gives me this output still no solution :(
Found sibling device [same parent]
Failed to set the LED of K4W or 1473 device: -1
[Stream 70] Negotiated packet size 1920
Failed to submit isochronous transfer 0: -1
Failed to submit isochronous transfer 1: -1
Failed to submit isochronous transfer 2: -1
Failed to submit isochronous transfer 3: -1
Failed to submit isochronous transfer 4: -1
Failed to submit isochronous transfer 5: -1
Failed to submit isochronous transfer 6: -1
Failed to submit isochronous transfer 7: -1
Failed to submit isochronous transfer 8: -1
Failed to submit isochronous transfer 9: -1
Failed to submit isochronous transfer 10: -1
Failed to submit isochronous transfer 11: -1
Failed to submit isochronous transfer 12: -1
Failed to submit isochronous transfer 13: -1
Failed to submit isochronous transfer 14: -1
Failed to submit isochronous transfer 15: -1
write_register: 0x0105 <= 0x00
write_register: 0x0006 <= 0x00
write_register: 0x0012 <= 0x03
write_register: 0x0013 <= 0x01
write_register: 0x0014 <= 0x1e
write_register: 0x0006 <= 0x02
write_register: 0x0017 <= 0x00
[Stream 80] Negotiated packet size 1920
Failed to submit isochronous transfer 0: -1
Failed to submit isochronous transfer 1: -1
Failed to submit isochronous transfer 2: -1
Failed to submit isochronous transfer 3: -1
Failed to submit isochronous transfer 4: -1
Failed to submit isochronous transfer 5: -1
Failed to submit isochronous transfer 6: -1
Failed to submit isochronous transfer 7: -1
Failed to submit isochronous transfer 8: -1
Failed to submit isochronous transfer 9: -1
Failed to submit isochronous transfer 10: -1
Failed to submit isochronous transfer 11: -1
Failed to submit isochronous transfer 12: -1
Failed to submit isochronous transfer 13: -1
Failed to submit isochronous transfer 14: -1
Failed to submit isochronous transfer 15: -1
write_register: 0x000c <= 0x00
write_register: 0x000d <= 0x01
write_register: 0x000e <= 0x1e
write_register: 0x0005 <= 0x01
write_register: 0x0047 <= 0x00

skribtsov, 2018-02-21:
I have the same issue, I really need this one to be solved, tried this on Odroid XU4 and Raspberry Pi. Same issue on all platforms. Have read about some hacks regarding the mystic "audio" build option, but haven't found a branch of libfreenect that would support it. No traces of audios.bin either in internet. PLEASE HELP!!!

dbautistav, 2018-07-29:
Some months late but I solved this issue by installing [proper drivers](https://github.com/shiffman/OpenKinect-for-Processing#kinect-v1-requirements). As my setup is Kinect 1 (1473) and win10/sdk v1.8, I used Zadig to install the **libusbK driver** for Kinect Audio, Xbox Camera and Kinect Security.

sener07, 2018-08-12:
I had the kinect for a while and i gave up trying after this problem but i am glad you find the solution, maybe i can look again, thank you very much, have a nice work