shiffman/OpenKinect-for-Processing #131: isochronous transfer error 4

TamaraDamceska, 2018-03-02:
I am trying to do the CameraParams example on kinect 1414 but I am getting this error isochronous transfer error 4 and I dont know how to fix, and I couldn't find anything on forums. Can you please help me

Grunewald17, 2018-03-11:
Same here with MacOS High Sierra. No RGB and no IR Image. Motor and LED are ok.

TamaraDamceska, 2018-03-12:
The problem was solve with the software zadig, that give u chance to install or replace softwares for the usb kinect (everything that is conected with SDK) witch are compatible to one on other. If you are downloading one by one you may have wrong ones. But this zadig will replace them. But I dont know if it's same for MacOS cause I am on windows. 
_I don't know if it is just me, but I am having a lot of problems with finding the way of working with kinect and processing, cause I'm not a programmer and no one has explained any issue in simple language on forums. For e.g. It took me one week to instal everything correctly and get the kinect connected with processing. Now I want to connect it with projector, and it would have been much easier if Shifman had tutorial on it, but someone Gene Kogan_ [](https://github.com/genekogan/KinectProjectorToolkit ) has a tutorial that starts when everything is perfect. I am starting to lose my nerves cause every time I solve a problem another one pops in. This Gene code won't draw the green screen , some programer told me it's cause of the graphics and processor are not separate, but how come that others are using it on laptops, not on desktop computer.Tomorrow I'll try on desktop with good graphic and If it doesn't work I don't know who to turn to :-/

hataxxi, 2018-07-05:
Using Processing 3.3.7 WIN10 64 with Kinect 1414. 
Im having exact same problems. No RGB and IR yet motor and led works fine.
I have downgraded zadig and tried reinstalling the drivers, i still get isochronous transfer error 0: -12 
Any other way of solving this error would be much appreciated.