shiffman/OpenKinect-for-Processing #133: There are no kinects, returning null

Michelleyaooooo, 2018-03-14:
Hi there,

I'm currently working on the OpenKinect-for-Processing library according to Daniel's tutorial. But I can't get the Kinect connected. Here's the error log in the console when running the point cloud from the library example:

_Found path: /Users/MichelleYAO/Documents/Processing/libraries/openkinect_processing/library/v1/mac/
Loaded freenect from /Users/MichelleYAO/Documents/Processing/libraries/openkinect_processing/library/v1/mac/libfreenect.dylib
No Kinect devices found.
Failed to open camera subdevice or it is not disabled.Failed to open audio subdevice or it is not disabled.There are no kinects, returning null_

I'm using a Mac OSX Serria (10.12.6) and my processing is 3.3.6.
Hardware: Kinect v1 (1414)

I've browsed through all possible solutions on Github and Processing Forum, tried (1)unplug and plug again for a thousand times (2)Try it on another computer 

There's other comments mentioned about USB 3.0 and 2.0. But not sure what it means since the mac I use only has USB3.0. I could found the device through system report(see the pic attached). For my laptop, I connected the Kinect with a power cable and a USB cable.

Please help;( Thank you so much in advance
[image]

totovr, 2018-03-15:
Try with this [SimpleOpenni](https://totovr.github.io/SimpleOpenni/)

Michelleyaooooo, 2018-03-26:
I tried it;( Still not working. But it works when I changed to Kinect v2.

marcoctn, 2018-06-18:
same problem here with kinect 1414, what can i try?

BrunoLerner, 2018-10-04:
same problem, with kinect 1414 in a Raspberry Pi 3

jasonamri, 2019-03-12:
Same problem with a 1414 on MacOS Mojave 10.14.3, I see the same thing in System Report, and get the same message in Processing

hamiltonbelinda0, 2020-02-02:
I had the same problem. I got it to work by uninstalling the Microsoft Kinect SDK and the Microsoft Kinect drivers. Used Zadig instead.

MuhammadAgni, 2020-09-30:
> I had the same problem. I got it to work by uninstalling the Microsoft Kinect SDK and the Microsoft Kinect drivers. Used Zadig instead.

Bismillah,..

Its Working
I thank you for it, and pray that God will bless you.