shiffman/OpenKinect-for-Processing #140: libusb error -12: LIBUSB_ERROR_NOT_SUPPORTED

itzku1, 2018-08-27:
Hi,
I'm using windows 10, kinect v2 with processing 3 and get this error:

64 windows
Init Kinect2 
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 17 usb devices connected
[Freenect2Impl] failed to open Kinect v2 @2:3!
[Freenect2Impl] found valid Kinect v2 @2:3 with serial 012920765147
[Freenect2Impl] found 1 devices
1 Device Connected!
[OpenCLDepthPacketProcessor::listDevice]  devices:
  0: GeForce GT 740M (GPU)[NVIDIA Corporation]
  1:       Intel(R) Core(TM) i7-3632QM CPU @ 2.20GHz (CPU)[Intel(R) Corporation]
  2: Intel(R) HD Graphics 4000 (GPU)[Intel(R) Corporation]
[OpenCLDepthPacketProcessor::init]  selected device: GeForce GT 740M (GPU)[NVIDIA Corporation]
Devce: 0
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 17 usb devices connected
[Freenect2Impl] failed to open Kinect v2 @2:3!
[Freenect2Impl] found valid Kinect v2 @2:3 with serial 012920765147
[Freenect2Impl] found 1 devices
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] closing...
[Freenect2DeviceImpl] deallocating usb transfer pools...
[Freenect2DeviceImpl] closing usb device...
**[UsbControl::claimInterfaces(IrInterfaceId)] failed! libusb error -12: LIBUSB_ERROR_NOT_SUPPORTED**
[Freenect2DeviceImpl] closed
[Freenect2DeviceImpl] failed to open Kinect v2 @2:3!
no device connected or failure opening the default one!

I tried libusbK drivers and any other drivers supported by Zadig and get this error or cannot recognize kinect at all.
I checked many forums and tutorials so far, no success.

Thanks for any help!

jeongcatfish, 2018-09-22:
same issue.
Did you solve it?

itzku1, 2018-09-22:
So try the library I mentioned above and save yourself some time.
If you insist and find any solution let me know!

jeongcatfish, 2018-09-22:
ThomasLengeling/KinectPV2 for processing
Do I have to download that library from github and add library myself?
Because there is no library at the "add library in processing".

itzku1, 2018-09-22:
download it from github and drop it in your processing-3.x\modes\java\libraries
I'm almost sure you can do everything you do with shiffman library with the other one including using the examples with minor changes (ThomasLengeling library has even more examples)

jeongcatfish, 2018-09-22:
Thank you for comment.
Do i have to change driver?

binarybabe33, 2018-11-20:
I have followed the instructions for the zadig usb driver install and it didn't work, so I want to switch back to my old drivers and use kinect pv2 like I could do yesterday, except I changed the drivers back in zadig and it doesnt work. it doesnt work in processing or max msp or in the kinect sdk..... did anyone successfully uninstall and revert the drivers?

itzku1, 2018-11-20:
Hi,
If you didn't find the answer yet, you can use windows device manager to
revert the drivers. Choose each kinect driver that you have changed and
choose to update the driver, then you'll have the option to choose from a
list so revert it to the default.

josebringas, 2019-11-28:
Thanks itzku1, your help is still relevant. Thomas Lengeling library works fine on my PC windows. Also i had to revert my driver to the default version as Zadig didn't work in my case. Cheers ppl!