shiffman/OpenKinect-for-Processing #141: kinect v1 PointCloud silhouette

mirceape, 2018-09-02:
Hi there,

Managed to get everything up and running with a kinect v1 1414 , latest edition of processing, windows8 and libusbK drivers. The PointCloud.pde works generates : 

1. Isochronous transfer error: 1, which doesn't seem to bother.
2. A dark silhouette thing, as seen in the attachement 

[image]

If anyone knows what causes it, i kindly ask for help.

Thx.

aptoptout, 2019-01-23:
I don't know about the Isochronous transfer error: 1 but the dark silhouette is just your shadow.

The Kinect projects infrared light on to everything in front of it. It's field of view is shaped like a pyramid. So by the looks of it, since you are sitting quite close and the infrared light rays that fall on you can't go through you but cast a shadow, there for the silhouette. Again this is showing maybe bigger than you since the rays come from the same origin and grow further away from each other the further they are cast.

So the silhouette isn't a bug or something going wrong, so I don't think there is any real issues here.

arturoherran, 2020-07-18:
Hi ! 

I hope you found the solution to this. Right now I am trying to find a way of disabling the IR camera from the v1 Kinect or at least modify the IR background pixels that create the shadow (or duplicate if you may), so If you happen to know how to achieve that, it would be great.