shiffman/OpenKinect-for-Processing #143: How can I remove libusbK driver??

jeongcatfish, 2018-10-05:
Hi I really enjoyed OpenKinect for Processing driver.
and now I woud like to use microsoft usb driver.
help me to remove libusbK...

binarybabe33, 2018-11-20:
I want to know also, my kinect v2 was working yesterday and now I try to use the drivers from zadig to run a script with openkinect and it doesn't work there or anywhere else now. I tried changing the drivers again in zadig to winusb and it didnt do anything...

elviktor, 2019-04-11:
I successfully removed the libusbK drivers for my Kinect v1.8 after some experimentation and exercising a lot of patience. I hope the following can help:

1. Delete the OpenKinect for Processing library.
2. Run a search for OpenKinect on your PC and delete any associated files that may come up.
3. Connect your Kinect to your PC.
4. Go to Windows Device Manager.
5. You should find a libusbK parent driver with a Kinect camera and audio driver as children.
6. Right click on the Kinect drivers and select 'Uninstall Device'.
7. If the drivers refresh than repeat the above step: right clicking and selecting 'Uninstall Device'.
8. Keep uninstalling until the drivers stop reappearing. (I had to repeat several times...)

If you are still unsuccessful:
1. Unplug the Kinect from your PC, reboot and then plug it back into the PC again to check if the drivers are there.
2. Run a search on your PC for 'libusbK' and delete as many files that come up.

I know the above process is kind of rough. However, I can say that I successfully deleted the libusbK driver and was able to reinstall the Windows SDK. Good luck!

smashah, 2020-01-23:
[image]

Make sure to tick the box to delete the driver while uninstalling. For further context, this is how I fixed the logitech unifying receiver on windows.

MartinAllende, 2022-05-05:
i have a ps4 joistick instaled by libusbk how can i solve this shit?

samanthajpetty, 2023-04-21:
I just signed in to say Thank you ...

closed 2023-04-21 by jeongcatfish