shiffman/OpenKinect-for-Processing #147: Library not Properly workin on Windos 10 and 8.1

JuanCristanchoVasquez, 2019-06-19:
Hi everybody! I'm having problems with the installation of the OpenKinect Library for Processing Library on both Windowds 10 Pro and 8.1 pro.  Inmediately after the library installation, running any sketch on the examples foler gives me the same error:

"A Library used by this Sketch is not installed properly"

Some of the errors on the console are:

"64 Windows
UnsatisfiedLinkError
xxx/xxxx/xxx/Processing/Libraries/openkinect_processing/Library/v1/mvsc/libusb1.0.dll
Can't find dependant libraries
A library relies on native code that's not available.
Or only works properly when the sketch is run as a 32-bit application."

Following previous issues I've tried changing the sketch path, manually install the library and reinstall everythin. By this point, I can say I've tried everything. 

First I thought it was a problem related with Windows 10 Pro, so I downgraded to Windows 8.1. At the begining it worked great but after some Microsoft remote assistence to activate the product in which the re installed de OS and downloaded various updates, I'm having the same "Library not Properly Installed" error again! After some really deep research I can say that it's a problem others have been experiencing. Some related threads are:

https://github.com/shiffman/OpenKinect-for-Processing/issues/74
https://github.com/shiffman/OpenKinect-for-Processing/issues/135

Please, there's been any update on this? It's turning a quite critical situation for me and the studio in which I work for

Thanks!

JMAndreas, 2019-06-26:
I get the same thing!

JuanCristanchoVasquez, 2019-06-26:
Hello again. We've finally solved this using Windows 8.1 and making sure to have installed all drivers. On Windows 10 pro 2019 we've never got to make it run. It would be great if someone with more deep technical background take a look at this.

Thanks!

nhungluu, 2019-06-27:
Hi Juan,
I was having the same issue and after researching for a while, it seems that Kinect has been discontinued by Microsoft a few years ago. It's not compatible with Windows 10. We're testing it with Windows 7 now.