shiffman/OpenKinect-for-Processing #150: There are no kinects, returning null

Phil-s-box, 2019-09-23:
Hi, i'm trying to set my kinect V1 1414 with Processing V3.
I'm using it on Windows 10 64 bits and i've already installed the drivers with zadig.

By the way it's working fine with Kinect Studio V1.8.

I have seen some issues with USB 3. This is the only port i have.

My problem is that i cannot find the camera when i'm launching Process.
Here's the response:

There are no kinects, returning null
Could not open camera: -12
Failed to open camera subdevice or it is not disabled. 
Failed to open motor subdevice or it is not disabled. 
Failed to open audio subdevice or it is not disabled.

Can you help me?

Regards

indirawrs, 2020-10-01:
I'm getting the same error. Did you ever work through it?

Phil-s-box, 2021-02-02:
It's been a long time but i think i simply used "windows SDK" to get the good drivers...

potun, 2021-05-02:
Hi Phil, I've been having the same issue. Please explain what you mean by 'used "windows SDK" to get the good drivers...'

maxtom92, 2023-06-03:
> It's been a long time but i think i simply used "windows SDK" to get the good drivers...

Hi excuse me, what do you mean by that?