shiffman/OpenKinect-for-Processing #153: Kinect Is Not Showing Depth Image

farhanfaheem, 2019-11-26:
I Am Getting This Error Can Anyone Help Me?
[image]

64 windows
UnsatisfiedLinkError: C:\Users\farha\Documents\Processing\libraries\openkinect_processing\library\v2\msvc\libusb-1.0.dll: Can't find dependent libraries
A library relies on native code that's not available.
Or only works properly when the sketch is run as a 32-bit application.
UnsatisfiedLinkError: C:\Users\farha\Documents\Processing\libraries\openkinect_processing\library\v2\msvc\libusb-1.0.dll: Can't find dependent libraries

alexgfaria, 2021-08-25:
I have the same problem. I'm on macOS Big Sur on an M1 chip.

stefanodalessio, 2024-02-07:
same here but only on Processing 4 (processing 3.5 all fine), M3 Sonoma