shiffman/OpenKinect-for-Processing #154: Window 10 Ruined the Fun.

systemprintln, 2020-01-06:
The library was working just fine a few months ago on Windows 10 but now it keeps throwing the same errors in the console:

'Can't find dependent libraries' 'A library relies on native code that's not available' 

Does someone has a fix? So sad to see support go, really had fun with the old Kinect (i'm using a 1414). Kinect4WinSDK is working but is much more limited in it's features.

systemprintln, 2020-01-06:
For those who wonder, I actually got it working:

1) Use Kinect SDK 1.7 (for model 1414)
2) Replace Drivers with Zadig (v2.0.1)
3) Manually install the openkinect library in the libraries folder

if Processing keeps returning 'There are no Kinects connected, returning null' 
4) open up task manager and end the process 'Kinect Management Service'
 
that did it for me.

kty, 2020-01-29:
@systemprintln Having the exact same problem, but your steps didn't work for me. :c

Could you share a little bit more about your environment?

- Which version of Windows 10 are you on? (I'm on Windows 10 Pro 64-bit)
- Which version of Processing 3? 64-bit or 32-bit?

I did all of your steps, but then I still have the above error with the libusb-1.0.dll:

64 windows
UnsatisfiedLinkError: [...]\Processing\libraries\openkinect_processing\library\v1\msvc\libusb-1.0.dll: Can't find dependent libraries
A library relies on native code that's not available.
Or only works properly when the sketch is run as a 32-bit application.
UnsatisfiedLinkError: [...]\Processing\libraries\openkinect_processing\library\v1\msvc\libusb-1.0.dll: Can't find dependent libraries
UnsatisfiedLinkError: [...]\Processing\libraries\openkinect_processing\library\v1\msvc\libusb-1.0.dll: Can't find dependent libraries

systemprintln, 2020-01-31:
@kty i'm on Windows 10 Home 64bit 1909 and using processing 3.5.3. Few notes:
- Are you sure you replaced ALL the Kinect devices drivers to libusbk? There should be an motor, camera, sound and security device showing up in Zadig. 
- Make sure you download the open-kinect package from github, not from the built-in library installer from Processing

systemprintln, 2020-01-31:
Also: have you checked which version of Kinect you have? The Kinect SDK 1.7 works fine with the 1414 but for other models using SDK 1.8 might be required.

justAguy321, 2021-11-09:
ive tried replacing all the drivers but i don't see a motor driver for v1 model #1437

fredciel, 2022-01-08:
Hi
I have 3 pc with three versions of windows 10.
On Windows Home 21H1, it does not work but it works on windows home 20H2 and windows Pro 1803.
I made all the recommendations on this page but it still doesn't work.
I have always the same error. Is it this new version of the system that is incompatible?

64 windows
java.lang.UnsatisfiedLinkError:
C:\Users\poesi\Documents\Processing\libraries\openkinect_processing\library\vl\msvc\libusb-1.0.dll:Cant find
dependent libraries
at processing.opengl. PSurfaceJOGL$2 .run (PSurfaceJOGL. java: 408)
at java.lang-Thread.run (Thread java: 748)
UnsatisfiedLinkerror:
C:\Users\poesi\Documents\Processing\libraries\openkinect_processing\library\v1\msvc\libusb-1.0.dll:Can'tfind
dependent libraries
A library relies on native code that's not available.
Or only works properly when the sketch is run as a 32-bit application.