shiffman/OpenKinect-for-Processing #156: Kinect 1473 - Mac OS X

cicekcocuk, 2020-04-14:
Hello everyone. I'm so new to coding so I hope you'll be able to help me with this.
I did everything and worked the libraries. But it always gives the no Kinect error with also blinking light.
I use a Kinect 1473 with Mac OS Mojave.
Anyone knows a workaround?
Best!

jspann, 2020-08-05:
What kinds of errors/exceptions are you getting? You might get the no Kinect error if your Kinect is not properly connected to your computer.

RobotLC, 2021-11-08:
Hi! I also got that error, however I'm working in ubuntu 20.03. I've suspected bad openni installation. Fortunately I've got another kinect, version 1414, and everything worked!

I suspect, for some reason that the library does not work with the 1473 version. After doing an lsusb, the output of the 1473 was:
Bus 001 Device 081: ID 045e:02ae Microsoft Corp. Xbox NUI Camera
Bus 001 Device 080: ID 045e:02ad Microsoft Corp. Xbox NUI Audio
Bus 001 Device 079: ID 045e:02c2 Microsoft Corp. Kinect for Windows NUI Motor

Instead of the output of the 1414:
Bus 001 Device 085: ID 045e:02ae Microsoft Corp. Xbox NUI Camera
Bus 001 Device 083: ID 045e:02b0 Microsoft Corp. Xbox NUI Motor
Bus 001 Device 084: ID 045e:02ad Microsoft Corp. Xbox NUI Audio

That's the only difference I've found.