shiffman/OpenKinect-for-Processing #165: Kinect data not showing on Apple Silicone M1 macOS 11.5.2

alexgfaria, 2021-08-25:
Seems like the data is getting through, but it isn't presented on the screen. I've tried adding frameRate(30), removing fullScreen(P3D), nothing works.

Please help!

hakimmazouz, 2021-09-23:
Did you find a way?

alexgfaria, 2021-09-24:
> Did you find a way?

yes. I installed processing 4.0b1 and processing-java through there. depth data now works, except live feed video.

ksenia-5, 2022-04-27:
**Apple Silicone M1 macOS 11.5.2 with Processing 4.0b8 in Java**
I get the error "The constructor Kinect(Kinect) is undefined", when running code from the RGB depth test found here
https://github.com/shiffman/OpenKinect-for-Processing/blob/master/OpenKinect-Processing/examples/Kinect_v1/RGBDepthTest/RGBDepthTest.pde

Any solution?