shiffman/OpenKinect-for-Processing #166: Object detection

finchle99, 2022-02-01:
Hi everyone,
I'm using Kinect v1 for detecting an object (a tomato in a tree). 
My idea is to use OpenCV to detect a tomato (I'm done with this step using Kinect's camera) and then I need to get the coordinates (x, y and z) of that tomato in a tree, I'm stuck in this step. Please help me with this problem. 
If there is something that goes wrong or impossible with my idea, please let me know and I would give many thanks to you if there is another method or idea that is easy to do for this project!
Thank you so much!

NickVerwoert, 2022-02-03:
Hi finchle, i was wondering if you ever found a solution, since we are trying to kinda achive the same thing, we need it to track a soccer ball.

finchle99, 2022-02-05:
Hi, I'm not done with this project..., but I got the depth data and the distance, but not really accurate much (about +- 4cm compare with reality).
If you still need a method from this step, you can try Pointcloud example from this library.

I'm using customed haar cascade for the detection step, but it's not good at all, but when I used Face haar cascade (from openCV library) and it's really good. This is the reference: https://www.youtube.com/watch?v=YX41KXbMf_U
If u have any good detection method, please help me.
Thank you.