shiffman/OpenKinect-for-Processing #168: M1 Mac OpenKinect Can't load library libusb-1.0.0.dylib

JoK-rgb, 2022-05-29:
I tried to run the DepthTest program on my M1 Mac but I ran into some issues. First Processing could not find the jna.jar and jna-platform.jar files. I solved this by copying them into the library folder. This seemed to work but now there is another issue. Processing can bot load the libusb-1.0.0.dylib library.

This is my error message: 
UnsatisfiedLinkError: Can't load library: /Users/UserName/Documents/Processing/libraries/openkinect_processing/library/v1/mac/libusb-1.0.0.dylib

How can I solve this?

aryaharditya, 2022-08-25:
i also have the same problem

pleekMan, 2022-10-24:
I'm on the same spot as you are...  :(

kjellxvx, 2022-11-11:
same issue, still no solution

fredciel2, 2022-12-28:
same for me too ...

maximocampo, 2023-02-07:
@JoK-rgb found the issue ?

j3nsykes, 2023-02-26:
If you download Processing 4.2 but the Intel version not silicon. Then run it via Rosetta it resolves this issue and M1 Mac with Kinect V1 works.  
libfreenet is not compatible with Silicon but is with Intel.

GuiFV, 2023-02-28:
Thanks @j3nsykes, It works! Tested here with kinect1 and 2. Running on M1!

ffd8, 2023-11-16:
> First Processing could not find the jna.jar and jna-platform.jar files. I solved this by copying them into the library folder.

Just tried to re-use Kinect on Processing 4.3 with a newer M1 mac... FYI for others that also land on this, the jna.jar + jna-platform.jar files can be found here: https://github.com/java-native-access/jna/tree/master/dist

Download the ones appropriate for your setup, just to be safe grabbed the following:
```
darwin-aarch64.jar
darwin-x86-64.jar
jna-platform-jpms.jar
jna-platform.jar
jna.jar
```

Drop them into the `openkinect_processing` / `library` folder. (also had to use the Intel version of Processing) – but it wooorked! thanks for the various tips above in this issue!

bustavo, 2024-10-12:
> > First Processing could not find the jna.jar and jna-platform.jar files. I solved this by copying them into the library folder.
> 
> Just tried to re-use Kinect on Processing 4.3 with a newer M1 mac... FYI for others that also land on this, the jna.jar + jna-platform.jar files can be found here: https://github.com/java-native-access/jna/tree/master/dist
> 
> Download the ones appropriate for your setup, just to be safe grabbed the following:
> 
> ```
> darwin-aarch64.jar
> darwin-x86-64.jar
> jna-platform-jpms.jar
> jna-platform.jar
> jna.jar
> ```
> 
> Drop them into the `openkinect_processing` / `library` folder. (also had to use the Intel version of Processing) – but it wooorked! thanks for the various tips above in this issue!
> 
> It's also a shame this library is no longer part of the Processing Library Manager? Tested removing then reinstalling- only to see it go pooof once removed and not be listed.. requiring a manual install.

This worked for me! Thanks!